pasko-zh/brzo_i2c #5: Not working with HTU21

loewexy, 2016-08-07:
I am failing to get your lib to work with the HTU21 Temperature/Humidity sensor. I am not quite shure if this is a problem with the lib or with my code. I always get a return code of 2 when finishing the transaction.

```c
#include "brzo_i2c.h"

unsigned char buf[10];

void setup() {

  Serial.begin(115200);

  Serial.println("Beginning application");
  
  brzo_i2c_setup(4,5,20000);

  brzo_i2c_start_transaction(0x80, 100);
  buf[0] = 0xE7;
  brzo_i2c_write(&buf[0], 1, true);
  brzo_i2c_read(&buf[1], 1, false);
  unsigned char returnCode = brzo_i2c_end_transaction();

  Serial.println(returnCode, HEX);
  Serial.println(buf[1], HEX);
}
```

pasko-zh, 2016-08-07:
My first impression is the following: 
According to the datasheet, the HTU21 has a fixed i2c address of 0x40: "_... I²C header consist of a 7-bit I²C device address 0x40 ..._" You can also find it in the [adafruit library](https://github.com/adafruit/Adafruit_HTU21DF_Library/blob/master/Adafruit_HTU21DF.h#L24).

But you are trying to talk to a slave at address 0x80 in `brzo_i2c_start_transaction(0x80, 100)`. You should use `brzo_i2c_start_transaction(0x40, 100)` instead. Note that [brzo_i2c_start_transaction](https://github.com/pasko-zh/brzo_i2c#i2c-transactions) takes the 7-bit slave address, you don't have to shift left the address by yourself, the library takes care of that.

Please let me know, if you were able to read out the User register of your HTU21.
paško

loewexy, 2016-08-07:
I should have mentioned that I tried both adresses 0x40 and 0x80 in fact I also built a loop cycling through all integers from 0 to 255 with no success.

pasko-zh, 2016-08-07:
I assume your hardware setup (pullups etc) is fine and the HTU21 works with the adafruit (wire) library, right? Currently I don't have an HTU21, so I cannot check. And I will be away from tomorrow on (holidays without IoT ;) but when I return, I will check.

loewexy, 2016-08-09:
The hardwaresetup should be correct. The Wire Libary did not work correct for me. I viewed the Signals on a scope and there were only two pulses. On on SDA one on SDL but it was no valid I2C. This is the reason I searched for another libary.

pasko-zh, 2016-08-18:
My [Adafruit HTU21 Breakout](https://www.adafruit.com/product/1899) just arrived. I did some first tests and the library **is** working. However, it seems that i2c timing with this board is something between OK and terrible :-/ 
I will post code and scope pics the next days.

closed 2016-10-05 by pasko-zh