pasko-zh/brzo_i2c #7: limited number of bytes?

Yoshi20, 2016-08-31:
First: Thanks a lot for your great work! :)

Currently I'm trying to generate some sound with the ESP8266 and a MCP4725 (DAC with a I2C inteface).
Since I would like to write a lot of data in one rush, i discovered that you made the following variable definitions in the write (and read) function:
void ICACHE_RAM_ATTR brzo_i2c_write(uint8_t *data, **uint8_t no_of_bytes**, boolean repeated_start)

My Question:
Is there a good reason you limited the number of bytes (no_of_bytes) to 256?
Could you increase that to at least an uint32_t?

I tested that localy with some random values and it seems to work fine so far. But you may be a better judge.

Keep your good work up :)

pasko-zh, 2016-09-01:
Thanks :v: 

Well, I just did not yet thought of the use case to send/receive more than 256 bytes in one write/read instruction. A first look in my code showed that it could be changed easily (I cannot test it now, but next week), so I will change this for 1.03 release.

However keep in mind: Your sketch should give the "esp8266 system" the opportunity to run from time to time. This is especially important when you need for instance wifi networking. Espressif's SDK documentation speaks of 15 msec, the arduino esp8266 community of 50 msec. Apart from this, in brzo_i2c reads and writes have all the interrupts disabled. Thus, single writes and reads should not take too long. See also the wiki, [maximum duration of reads and writes](https://github.com/pasko-zh/brzo_i2c/wiki#atomic-reads-and-writes) or [clock stretching](https://github.com/pasko-zh/brzo_i2c/wiki#i2c-setup-and-clock-stretching).
Therefore, depending on your use case, it might work well to send some big chunk of bytes at once, or you would neet to split it up into smaller chunks and use either `yield()` or `delay(.)` between consecutive reads/writes.

btw: The MCP4725 supports fast mode up to 400 kHz SCL speed. But since it is a i2c high speed device, it might work to increase fast mode speeds up to 800 or even 1000 kHz (i.e. fast mode *plus* speeds). But then you need to decrease the value of the pull-up resistor down to 1 k ohm. If you have a scope, check the timings.

pasko-zh, 2016-09-18:
Added to Release [1.1.0](https://github.com/pasko-zh/brzo_i2c/releases/tag/1.1.0)

closed 2016-09-18 by pasko-zh