pasko-zh/brzo_i2c #12: Question about "round" method

piersfinlayson, 2016-09-21:
Hi - nice library!  Making use of it in my non-RTOS esp SDK application the linker is not able to find the round() function in any of the system libraries I'm already including.  Where are you expecting to pick this up from?  I'm already having problems with available RAM so I'm keen not to include any extra library functions unnecessarily!

valkuc, 2016-09-21:
About round see my last comment here https://github.com/pasko-zh/brzo_i2c/pull/10

piersfinlayson, 2016-09-21:
@valkuc - Yes, makes sense to me.  I'll use your workaround for now.

pasko-zh, 2016-09-21:
Of course @valkuc is absolutely right! No floating point operations are needed to perform the round as I wanted it to have. Don't know why I didn't catch that much earlier, I guess I am getting old :-/ 

I will change this in the next release.

valkuc, 2016-09-22:
"Two heads are better than one"
so, no problem ;-) I'm not young too :-)

pasko-zh, 2016-10-05:
Added to 1.2.0

closed 2016-10-05 by pasko-zh