pasko-zh/brzo_i2c #22: Enabling -Wall GCC option shows some warnings

valkuc, 2017-10-08:
I'm not familar with assembler, please check either those variables are used or can be safely removed:
```
'a_bit_index' is used uninitialized in this function [-Wuninitialized]	brzo_i2c.c	line 768
'a_bit_index' may be used uninitialized in this function [-Wmaybe-uninitialized]	brzo_i2c.c	line 73
'a_bit_index' may be used uninitialized in this function [-Wmaybe-uninitialized]	brzo_i2c.c	line 389
'a_in_value' is used uninitialized in this function [-Wuninitialized]	brzo_i2c.c	line 768
'a_in_value' may be used uninitialized in this function [-Wmaybe-uninitialized]	brzo_i2c.c	line 73
'a_in_value' may be used uninitialized in this function [-Wmaybe-uninitialized]	brzo_i2c.c	line 389
'a_set' is used uninitialized in this function [-Wuninitialized]	brzo_i2c.c	line 768
'a_set' is used uninitialized in this function [-Wuninitialized]	brzo_i2c.c	line 1082
'a_set' may be used uninitialized in this function [-Wmaybe-uninitialized]	brzo_i2c.c	line 73
'a_set' may be used uninitialized in this function [-Wmaybe-uninitialized]	brzo_i2c.c	line 389
'a_temp1' is used uninitialized in this function [-Wuninitialized]	brzo_i2c.c	line 768
'a_temp1' is used uninitialized in this function [-Wuninitialized]	brzo_i2c.c	line 1082
'a_temp1' may be used uninitialized in this function [-Wmaybe-uninitialized]	brzo_i2c.c	line 73
'a_temp1' may be used uninitialized in this function [-Wmaybe-uninitialized]	brzo_i2c.c	line 389

```

pasko-zh, 2017-10-15:
These variables are only needed for the inline assembler. They are on the "[clobber list](https://github.com/pasko-zh/brzo_i2c/blob/master/brzo_i2c.c#L354)": This means, you can set an alias variable name that you can use in the inline assembly. For instance, `a_temp1` can be referred to as `r_temp1` in the inline assembly. 

The "downside" is that if you want to use variables in inline assembly instead of registers, you have to do it this way. And the compiler may give you these warnings, because in the rest of the C code these variable are not used. 

The advantage is that the compiler will then replace these variables, e.g., `r_temp1`, in an optimal way with real registers.

valkuc, 2017-10-15:
Ok, no problem!

closed 2017-10-15 by valkuc