pasko-zh/brzo_i2c #23: Implementing I2C slave for MAX11615 not working as expected

w0w, 2017-11-15:
I am implementing i2c communication with MAX11615 and i am not getting desired result. Below is a snapshot to give more clarity on the slave. [Datasheet](https://datasheets.maximintegrated.com/en/ds/MAX1615-MAX1616.pdf)
[image]

My code looks like 

```#include <brzo_i2c.h>
uint8_t SDA_PIN = 4;
uint8_t SCL_PIN = 5;
uint8_t MAX11615_ADR = 0x33;
uint8_t buffer[8];
uint8_t error = 0;
float pH = 0.0;
float EC = 0.0;
uint8_t ICACHE_RAM_ATTR get_pH_EC(float *p, float *e) {
  uint32_t a_code = 0;
  uint32_t b_code = 0;
  uint32_t c_code = 0;
  uint32_t d_code = 0;
  uint8_t bcode = 0;
  brzo_i2c_start_transaction(MAX11615_ADR, 400);
  buffer[0] = 0xD2; 
  buffer[1] = 0x09;
  
  brzo_i2c_write(buffer, 2, false); //writes 2 bytes to slave address, repeated start is false
  brzo_i2c_read(buffer, 8, false); // read 8 bytes to slave address, repeated start is false
  bcode = brzo_i2c_end_transaction();
  delay(50);
  a_code = ((buffer[0] << 4) | buffer[1]);
  b_code = ((buffer[2] << 4) | buffer[3]);
  c_code = ((buffer[4] << 4) | buffer[5]);
  d_code = ((buffer[6] << 4) | buffer[7]);
  *p = a_code;
  *e = b_code;
  Serial.print(" | a = "); Serial.print(a_code);
  Serial.print(" | b = "); Serial.print(b_code);
  Serial.print(" | c = "); Serial.print(c_code);
  Serial.print(" | d = "); Serial.print(d_code); 
  return bcode;
}
void setup() {
  delay(1000);
  Serial.begin(115200);
  brzo_i2c_setup(SDA_PIN, SCL_PIN, 2000);
}
void loop() {
  Serial.println();
  Serial.println("Waiting 5 seconds...");
  delay(5000);
  error = get_pH_EC(&pH, &EC);
  if (error == 0) {
    //Print ph and ec
  }
  else {
    Serial.print("Brzo error : ");
    Serial.println(error);
  }
} ```

What is tripping me is the values are same for a_code, b_code, b_code, d_code.  The setup and configuration byte is proper and checked with manufacturer. Is it possible that the write is not happening correctly and hence giving value from one channel, I am not very versed with writing assembly to not sure how to debug this. 

Thanks

pasko-zh, 2017-11-15:
I will have a look at it, as soon as "my other work" let me to do so ;-)

pasko-zh, 2017-11-21:
Some questions from my side: 
1. What is the size of your pullup resistor on SDA and SCL? 
2. Which esp-board are you using?
3. Which version of Arduino IDE and [esp8266 core](https://github.com/esp8266/Arduino/releases) do you use?
4. You don't get any `brzo_i2c` errors, right?
5. Do you get back different values, or always the same, independant of what happens on the analog inputs of the MAX11615?
6. Did you try different settings for status and configuration register? Did it work?
7. Does i2c communication work in general in your setup, i.e. were you able to try it out with a different ic2c slave? Or is the problem only occuring with the MAX11615.
5. Do you have a scope or logical anaylizer?

I had a (quick) look at the datasheet (wrong link in your comment, but don't worry, I found the correct one). AFAIK, there is no way to read out the configuration or status register, right? Otherwise, we could have checked that the values were written correctly (assuming there are no read errors).
Even if the the write, i.e. bits 6 and 5, was not successful (i.e. writing 0), the power-up values of these would be 0 0, so the SCAN selection mode should not be affected.

w0w, 2017-11-28:
1. What is the size of your pullup resistor on SDA and SCL?
`4.7k `

2. Which esp-board are you using?
`esp8266 nodemcu 1.0`

3. Which version of Arduino IDE and esp8266 core do you use?
`Latest`

4. You don't get any brzo_i2c errors, right? 
`No brzo errros`

5. Do you get back different values, or always the same, independant of what happens on the analog inputs of the MAX11615?

`I have figured out a way to work with wire as of now but i want to use brzo and i am trying to understand more about the code, if you can point me to any reading material it will be great.`

6. Did you try different settings for status and configuration register? Did it work?

`I think it worked and i have narrowed down the error to bit shifting and noise from the power supply. `

7. Does i2c communication work in general in your setup, i.e. were you able to try it out with a different ic2c slave? Or is the problem only occuring with the MAX11615.

`I don't have any other i2c setup as of now.`

Do you have a scope or logical anaylizer?

`Not yet, i am ordering one soon.`

It must be a stupid question to ask but what is the console.log equivalent in assembly ?

closed 2017-11-28 by w0w

pasko-zh, 2017-12-05:
Thanks for additional info. Just a small remark: I have not yet tested `brzo_i2c` with the latest Arduino core , i.e., 2.4-rc*. But normally this is not a problem... 

So, if I understood you correctly, i2c communication is working?

Concerning assembly language question (which is not stupid question at all): Well, debugging is always a bit difficult with these microcontrollers. It strongly depends on the IDE you are using, how good or bad it is supported. This is also true for (inline) assembly language. At least from the docs, the [Xtensa Xplorer IDE](https://ip.cadence.com/swdev) seems to give good support also for assembly, i.e. you can observe the contents of the registers etc; but I am not using it.

Otherwise, since you are on the instruction level with (inline) assembly, the only way to go is to think and think and think again before implementing something ;-) ....and to "debug" like this: If possible, split up inline assembly in parts, pass values from register to the outside world (i.e. c/c++) and then print them to the console. 

A help for me to debug `brzo_i2c` was the rigol scope: I could know from the observed signals, where in the assembly code something probably went wrong.