pasko-zh/brzo_i2c #26: Question:  Logic Levels and  ...

Eddiiie, 2018-01-31:
Hello,

I have a question.   ESP8266 is using 3.3v on the GPIO pins, right?
I am assuming most I2C devices are 5v.   
Are you assuming a logic level converter is in use or is there no issue between the 3.3 and 5v.
OR am I just wrong and all I2C is 3.3v?

Unfortunately, I forgot the second question.  grrr

pasko-zh, 2018-02-03:
Hi Eddiie,

1. Yes, the esp8266 has 3.3V on the pins
2. Well this depends, up to now I have found all i2c devices that I need with 3.3V :smile:
3. Most breakout boards with i2c devices include already a logic level shifter, e.g., [Adafruit's HTU21 Breakout Board](https://www.adafruit.com/product/1899). 
However, be aware that those boards may have a built-in pull-up resistor with a very high resistance (in i2c jargon this is referred to as "weak pull-up", because the current flowing through is very weak). Such a high resistor will leave you with terrible raise times, so a smaller value ("stronger" pull-up) is needed. See my comments for [HTU21D with brzo_i2c.](https://github.com/pasko-zh/brzo_i2c/tree/master/examples/HTU21D)

Eddiiie, 2018-02-06:
Thank you for all of that information.  I am working with a device that is most likely expecting 5v levels.  It is not working currently with the ESP8266, will try adding level shifter and see, maybe do a capture to see what the bus looks like.