pasko-zh/brzo_i2c #27: Cannot compile for Heltec Wifi Kit 32

Eddiiie, 2018-02-14:
Hello,

Today I get compile error.  Can you please confirm if this lib works ok for Helteck Wifi 32.
Meanwhile, I will test sample code.

Thank you!

..........................
WARNING: library Brzo_I2C claims to run on (esp8266) architecture(s) and may be incompatible with your current board which runs on (esp32) architecture(s).
..........................
libraries\Brzo_I2C\brzo_i2c.c.o:(.iram1.literal+0x2c): undefined reference to `system_get_cpu_freq'

libraries\Brzo_I2C\brzo_i2c.c.o:(.literal.brzo_i2c_setup+0x14): undefined reference to `system_get_cpu_freq'

libraries\Brzo_I2C\brzo_i2c.c.o: In function `brzo_i2c_start_transaction':

C:\Users\IlikeB00bs\Documents\Arduino\libraries\Brzo_I2C/brzo_i2c.c:989: undefined reference to `system_get_cpu_freq'

libraries\Brzo_I2C\brzo_i2c.c.o: In function `brzo_i2c_setup':

C:\Users\IlikeB00bs\Documents\Arduino\libraries\Brzo_I2C/brzo_i2c.c:1044: undefined reference to `system_get_cpu_freq'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Heltec_WIFI_Kit_32.

Eddiiie, 2018-02-14:
Same for board Error compiling for board "Heltec_WIFI_LoRa_32".   
Expected.

Code compiles ok when selecting NodeMCU type, but will not upload obviously.

cry

pasko-zh, 2018-02-17:
The library is written in (inline) assembly and hence does **only** work on the esp8266 it will **not** work for esp32.
Note that the esp32 has hardware support for i2c, thus implementing i2c in software wouldn't make much sense.

Eddiiie, 2018-02-20:
Oh, sorry about that.    I am going too fast.  :)

closed 2018-02-20 by Eddiiie