pasko-zh/brzo_i2c #29: Multiple I2C

cmoine, 2018-04-20:
I haven't find a way to contact the author of this program, so I fill a request:

I am using an ESP8266 (but it would be an Arduino I'd think it would be the same) to read temperature from BME280 (I2C Master), but I'd like to add I2C Slave capabilities on the same ESP to communicate with another device (Slave is a strong constraint). I read the source code of this library, and I don't see any reason why it could be feasible:
- This is software I2C
- I would like to act like a I2C server: when I receive a request from the master, I read temperature as an I2C master.

You'll probably ask yourself why my ESP & BME280 aren't slave with different addresses ?

Well there are several reasons that makes me want continue on this solution:
- There are several BME280, which requires an I2C multiplexer.... Coupling Multiplexer for BME280 and no mutiplexer for ESP seems for me a bit... dirty (I can live with it though)
- Some BME280 are not always working properly, I have no idea whether it comes from the mutliplexer, or the wires. I tried some things, but for me it looks strange, it depends on the order of the BME on the multiplexer, or something strange like that.
- I'd like to have a much stronger approach of the entire installation: I'd like to have entirely independent component (component would consist of ESP + BME + other sensors), with different I2C address. Either the component works, either it's not, and that's it. I can test components individually, and consider them as a whole from the master perspective.

Why don't I use Wifi ? Because it does, but my wife doesn't want Wifi for children 😀 

I would highly appreciate any suggestion, or if you think it is a valid use case like me, and think it is feasible quite easily, I can provide a PR (if you are interested of course).

Thank you in advance,

Cheers.

pasko-zh, 2018-05-04:
Hello!

If you have questions you can either ask them here, or contact me via PM. I prefer the former, so that also others can read question and (my) answers.

So, I am trying to understand your question.

Usually, one is connecting sensors as slaves to the i2c bus and the esp8266 has the role af an i2c master, i.e. the i2c master. 

You can only connect up to two BME280 on the same i2c bus, since according to the [datasheet ](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME280_DS001-12.pdf)page 31, you can only set LSB of the slave address, resulting in either 0x76 or 0x77. 

So, I am bit confused when you write "to read temperature from BME280 (I2C Master)", since a BME280 cannot be an i2c master, it only supports being an i2c slave. 

1. Would you like to have several i2c buses with one esp8266, such that you can use more than two BME280?
2. When you write "Some BME280 are not always working properly", do you use some BME280 breakout boards, e.g., from [Adafruit](https://learn.adafruit.com/adafruit-bme280-humidity-barometric-pressure-temperature-sensor-breakout/overview)? 
And what is the size of your i2c pullup resistors? 
Which i2c multiplexer are you using?
Does the problem persist, when you connect the BME280 without the i2c multiplexer, i.e. directly to the i2c?

cmoine, 2018-12-05:
Hi, I am very sorry for the very late reply.

Let me rephrase what I would like to do (not sure whether it good idea, but anyway, it is not a all working anymore with the I2C Multiplexer):
- I have a Raspberry that discuss with several Arduino slave using I2C. So far so good.
- The Arduino (Nano) is responsible for commanding relay, reading a sensor to know if the window is open, AND I would like to read BME280 values.

The Arduino Nano is already an I2C slave, so it cannot additionnally be a I2C master for the BME280 sensor.

Why not puting the BME on the same I2C with different address ? Becasue I have several Arduino, and I want to keep them isolated, without the I2C Multiplexer which doesn't work :/ Or maybe it is Raspberry PI + Multiplexer that doesn't work properly, no clue....

Why not using SPI ? just because my 4 BME280 sensors have only I2C pins exposed :/

Problem with newer version of BRZO? it doesn't work anymore with Arduino:
```WARNING: library Brzo_I2C claims to run on (esp8266) architecture(s) and may be incompatible with your current board which runs on (avr) architecture(s).
C:\Users\Christophe Moine\Documents\Arduino\libraries\Brzo_I2C\brzo_i2c.c:52:22: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'brzo_i2c_write'

 void ICACHE_RAM_ATTR brzo_i2c_write(uint8_t *data, uint32_t no_of_bytes, bool repeated_start)

                      ^

C:\Users\Christophe Moine\Documents\Arduino\libraries\Brzo_I2C\brzo_i2c.c:367:22: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'brzo_i2c_read'

 void ICACHE_RAM_ATTR brzo_i2c_read(uint8_t *data, uint32_t nr_of_bytes, bool repeated_start)

                      ^

C:\Users\Christophe Moine\Documents\Arduino\libraries\Brzo_I2C\brzo_i2c.c:752:22: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'brzo_i2c_ACK_polling'

 void ICACHE_RAM_ATTR brzo_i2c_ACK_polling(uint16_t ACK_polling_time_out_usec) {

                      ^

C:\Users\Christophe Moine\Documents\Arduino\libraries\Brzo_I2C\brzo_i2c.c:989:22: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'brzo_i2c_start_transaction'

 void ICACHE_RAM_ATTR brzo_i2c_start_transaction(uint8_t slave_address, uint16_t SCL_frequency_KHz)

                      ^

C:\Users\Christophe Moine\Documents\Arduino\libraries\Brzo_I2C\brzo_i2c.c:1023:25: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'brzo_i2c_end_transaction'

 uint8_t ICACHE_RAM_ATTR brzo_i2c_end_transaction()

                         ^

C:\Users\Christophe Moine\Documents\Arduino\libraries\Brzo_I2C\brzo_i2c.c:1041:24: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'brzo_i2c_setup'

 void ICACHE_FLASH_ATTR brzo_i2c_setup(uint8_t sda, uint8_t scl, uint32_t clock_stretch_time_out_usec)

                        ^

C:\Users\Christophe Moine\Documents\Arduino\libraries\Brzo_I2C\brzo_i2c.c:1105:24: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'brzo_i2c_reset_bus'

 void ICACHE_FLASH_ATTR brzo_i2c_reset_bus()

                        ^

exit status 1
Error compiling for board Arduino Nano.```