pasko-zh/brzo_i2c #33: AM2320 sensor not found

v-a-d-e-r, 2018-11-06:
Hi. With wire library the sensor is found.

pasko-zh, 2018-11-06:
Well _vader_ ... your thoughts are strong, very strong and poweful, but you still need a [tray](https://www.youtube.com/watch?v=Sv5iEK-IEzw) 😄 
i.e. you would have to show me your code, otherwise I could only think what you might are doing...

v-a-d-e-r, 2018-11-06:
Well, I was changing the orig. wire library with yours in ESPeasy. There is a scan function which looks like this:
  for (address = 1; address <= 127; address++ )
  {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0)
   ...bla bla....

with this:
  for (address = 1; address <= 127; address++ )
  {
    brzo_i2c_start_transaction(address, 100);
    brzo_i2c_ACK_polling(1000);
    error = brzo_i2c_end_transaction();
    if (error == 0)
...bla bla....

The orig. code works and find all sensors, yours find all, except the AM2320. For what reason ever.... :-/

pasko-zh, 2018-11-25:
I guess it is some timing issue... probably the AM2320 is not 100% within the i2c spec (`brzo_i2c` follows them very strictly). I will order one and do some checks with the scope.

v-a-d-e-r, 2018-11-26:
From the data sheet:

Step one: Wake Sensor
In order to reduce the humidity sensor errors caused by self-heating, the sensor in a non-working state, dormant, so to read the sensor must wake sensor to transmit commands to read and write, otherwise the sensor will not respond. It should be noted that, in the wake sensor, I2C address is sent, the sensor does not respond to ACK, but the host must send back an ACK clock verify that the ninth SCL clock signal. Wake sensor operation is following instructions:
After the host sends a start signal is applied to the starting address, wait for sometime (waiting time of at least 800 μs, the maximum 3ms; such as host hardware I2C, you do not need to wait for the hardware I2C will automatically wait), and then sends a stop signal.
Namely: the starting signal +0xB8 + wait (> 800us) + stop signal timing diagram shown in Figure 15....

Maybe the >800µs is the problem...???

pasko-zh, 2018-11-26:
Well, I've already tried to go through this chinglish datasheet... just horrible :-/
Fig 15 looks like i2c clock stretching, but it isn't, since it is supposed to occur _before_ the 9th clock cycle and not after... 

Btw, I just saw that you are using only `brzo_i2c_ACK_polling(.)`, you should only use this it if you have a slave needing it, e.g., EEPROMS. Just use `brzo_i2c_write(.)`. Maybe to wake-up the device you need to have two complete write transactions. 

But with this kind of datasheet and without a scope it's shooting in the dark...

v-a-d-e-r, 2018-11-26:
Yes, the datasheet is a bit scary. They are talking about an I2C address of 0xB8, but it is 0x5C. :-/ Ok, I'll try with brzo_i2c_write(.) ....

v-a-d-e-r, 2018-12-28:
Hi. Why is this not working?
brzo_i2c_start_transaction(address, 100);
error = brzo_i2c_end_transaction();
only this...
brzo_i2c_start_transaction(address, 100);
brzo_i2c_ACK_polling(1000);
error = brzo_i2c_end_transaction();

You wrote, that brzo_i2c_ACK_polling should normally not be used. So what I'm doing wrong?

BTW: Did you get the sonsor?

v-a-d-e-r, 2018-12-29:
Hi. Why is this not working?
brzo_i2c_start_transaction(address, 100);
error = brzo_i2c_end_transaction();
only this...
brzo_i2c_start_transaction(address, 100);
brzo_i2c_ACK_polling(1000);
error = brzo_i2c_end_transaction();

You wrote, that brzo_i2c_ACK_polling should normally not be used. So what I'm doing wrong?

BTW: Did you get the sensor?

pasko-zh, 2018-12-31:
The sensors are on their way from China to Switzerland, as I said I will then check with my scope what is going on.

Hansen-db, 2019-01-06:
I had the same problem with AM2320, but solved it by adding another pair of begindTransmission / endTransmission.
It seems that this solution has also been discussed at https://forum.arduino.cc/index.php?topic=399437.15

My solution:
  Wire.beginTransmission(_addr); 
  delayMicroseconds(900); //>0.8ms
  Wire.endTransmission(); // The AM2320 is now awake
  Wire.beginTransmission(_addr); // line added as a hack 
  Wire.endTransmission();  // line added as a hack

v-a-d-e-r, 2019-02-04:
Ah, ok. Thanks for clearing this. But I kick the crappy sensor into trash, that goes by far faster than writing a special code for it... ;-)