pasko-zh/brzo_i2c #39: Novice obsession

zhubinsheng, 2020-01-06:
你好，很抱歉打扰你，希望你在百忙之中帮我分析下困扰我已久的问题：
在使用您的库之后，存储和读取操作的返回结果都是0，读取是有结果的，但是存储好像并没有执行，只是偶尔会执行一次。我是用esp8266和24c16通讯

Hello, I'm sorry to disturb you. I hope you can help me analyze the problems that have been bothering me for a long time in your busy schedule:

After using your library, the return results of the storage and reading operations are both 0, and the reading has results, but the storage does not seem to execute, only occasionally. I use esp8266 to communicate with 24C16

pasko-zh, 2020-01-08:
I will have a look at it, but please allow some time for my replies because of my "normal" work ;-)

zhubinsheng, 2020-01-16:
Hello, I'm sorry to disturb you before. It's my problem. I found the reason and solution after referring to the following website. Thank you very much
https://github.com/P-Find/arduino-24c16/blob/master/_24c16/_24c16.ino

closed 2020-01-16 by zhubinsheng